NiryoRobotics/ned_ros #5: [niryo_robot_arm_commander-9] process has died

robbenv, 2021-09-28:
Im a learner of this product,and I want to run an example by **Python ROS Wrapper** in **rviz**, according to [documentation].(https://docs.niryo.com/dev/ros/v3.2.0/en/source/python_ros_wrapper/examples_basics.html#your-first-move-joint)
but when I executed `python3 move_joints.py`(an example copied from the documentation), I got the flowing error:  

> niryo_robot_python_ros_wrapper.ros_wrapper.NiryoRosWrapperException: Action Server is not up : /niryo_robot_arm_commander/robot_action

Does it mean that I should run **"Action Server"** by executing `python robot_commander_node.py` before I want to run any example?

But I also found when i start up rviz by `roslaunch niryo_robot_bringup desktop_rviz_simulation.launch`, a flowing error will come up:

> **niryo_robot_arm_commander-9] process has died** [pid 45506, exit code 1, cmd /home/luwei/code/catkin_ws_niryo_ned/devel/lib/niryo_robot_arm_commander/robot_commander_node.py __name:=niryo_robot_arm_commander __log:=/home/luwei/.ros/log/015601d4-201f-11ec-a2f3-cf3e78bbc405/niryo_robot_arm_commander-9.log].
log file: /home/luwei/.ros/log/015601d4-201f-11ec-a2f3-cf3e78bbc405/niryo_robot_arm_commander-9*.log
[niryo_robot_poses_handlers-11] process has died [pid 45514, exit code 1, cmd /home/luwei/code/catkin_ws_niryo_ned/devel/lib/niryo_robot_poses_handlers/poses_handlers_node.py __name:=niryo_robot_poses_handlers __log:=/home/luwei/.ros/log/015601d4-201f-11ec-a2f3-cf3e78bbc405/niryo_robot_poses_handlers-11.log].

is it the reason why "Action Server" didn't run?

my environment:
Ubuntu20.04
ROS-noetic
I also attached the log file of starting up rviz
[roslaunch-baiyu-44368.log](https://github.com/NiryoRobotics/ned_ros/files/7241443/roslaunch-baiyu-44368.log)
.

ValentinPitre, 2021-09-28:
Unfortunatly, Ned stack only works on ROS-melodic with Python 2.7. If you want to use ROS Melodic on Ubuntu 20.04, you have to build ROS from source http://wiki.ros.org/melodic/Installation/Source :)

robbenv, 2021-09-28:
> Unfortunatly, Ned stack only works on ROS-melodic with Python 2.7. If you want to use ROS Melodic on Ubuntu 20.04, you have to build ROS from source http://wiki.ros.org/melodic/Installation/Source :)

thanks for your reply, do you have any plan to support Ubuntu20.04 and ROS-Noetic in the future?

jmottier-niryo, 2023-10-20:
Hello, sorry for the (very) late reply.
Starting the 5.0.0 version, the ned2 is running under Ubuntu 20.04 and ros noetic.

closed 2023-10-20 by jmottier-niryo