NiryoRobotics/ned_ros #7: Need some suggestions to use Niryo Ned with ROS2?

nqduy35, 2022-02-21:
Hello, do you plan to develop a ROS2 version for Niryo Ned? I tried to control and connect Niryo Ned with other robots running ROS2. Do you have any suggestions?

ValentinPitre, 2022-03-02:
We are thinking about it. But I can't confirm anything. 

A simple way to interconnect a ros2 robot with a Ned is to use the python api pyniryo to control the robot remotely.

codesim, 2022-10-19:
Any news on this?
Using pyniryo, I would have to build a ROS2 node defining some glue logic between the pyniryo API and ROS2 services/messages.  Do you plan to make this easier by providing native use of ROS2?

Jpl-niryo, 2022-11-15:
Hi,

At some point we will transition over to ROS2 but our latest update will first be to switch over to ROS Noetic. It is hard to tell when we will move over to a native use of ROS unfortunately

nqduy35, 2022-11-15:
Thank you for the information. It is also very nice if the Niryo team could migrate the code from ROS 1 to ROS 2 using ros1_bridge (https://github.com/ros2/ros1_bridge)

closed 2023-10-20 by jmottier-niryo