NiryoRobotics/ned_ros #8: Can not change Modbus port from 5020 to 502

Thuunderbear, 2022-04-13:
Hello,
This bug occur on ned one.
In "~/catkin_ws/src/niryo_robot_modbus/launch/modbus_server.launch" when changing :

> \<arg name="server_port" default="5020"/>

as recommended to

> \<arg name="server_port" default="502"/>

nothing happen after restarting ( I check the port using nmap and nothing has change).

After some research I reach the niryo_robot_modbus package folder by using :

> roscd niryo_robot_modbus

it send me to "~/catkin_ws/install/release/ned/share/niryo_robot_modbus/launch" folder and when I try to change the same file in this folder after this commands :

> rosdep fix-permissions

> roslauch niryo_robot_modbus modbus_server.launch

Thes modbus TCP port is closed and I have an error

> ModbusServer.init : TCP server unable to start : [Errno 13] Permission denied

(sudo seems to doesn't work )

Could you help me ?

Thanks

ValentinPitre, 2022-04-13:
Port 502 does not seem to be available. Have you tried with another port? 1265 for example? 

Editing both _"~/catkin_ws/src/niryo_robot_modbus/launch/modbus_server.launch"_ and _"~/catkin_ws/install/release/ned/share/niryo_robot_modbus/launch"_ is the correct method.
You can also simply edit the first file and run `catkin build` to automatically update the second.

Thuunderbear, 2022-04-13:
> Port 502 does not seem to be available. Have you tried with another port? 1265 for example?

You are right, I can change Modbus port to an other port except 502. Is there any way to put ModbusTcp on port 502? I need this port to communicate with an other device.

ValentinPitre, 2022-04-14:
I've tryed to launch the ros node with sudo, it seems working.

In the both .launch files change the 12th line by:
`<node name="niryo_robot_modbus" pkg="niryo_robot_modbus" type="modbus_server_node.py" output="screen" launch-prefix="echo robotics | sudo -S">`

Thuunderbear, 2022-04-21:
Thanks a lot for your answer. I found another solution using this command:

> sudo socat TCP-LISTEN:502,fork TCP:localhost:5020

and I created a service to launch this command when the raspberry is booting. I haven't try your solution but maybe someone will need your solution.

closed 2022-04-21 by Thuunderbear