NiryoRobotics/ned_ros #9: error launching roslaunch niryo_robot_bringup desktop_gazebo_simulation.launch - hardware_interface::EffortJointInterface'. Make sure this is registered in the hardware_interface::RobotHW

postak, 2022-07-04:
When I start the ROS using the command

`error launching roslaunch niryo_robot_bringup desktop_gazebo_simulation.launch`

I get these errors

```
[ERROR] [1656922485.651503292, 5.555000000]: This controller requires a hardware interface of type 'hardware_interface::EffortJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
[ERROR] [1656922485.651573723, 5.555000000]: Initializing controller 'gazebo_tool_commander' failed
[ERROR] [1656922486.653018, 6.219000]: Failed to load gazebo_tool_commander
```

Any suggestions?

postak, 2022-07-04:
it seems that libgazebo_ros_joint_state_publisher.so is missing but I don't know where to get it (from which package)

ValentinPitre, 2022-08-11:
sudo apt install ros-noetic-gazebo-ros-control??

postak, 2022-08-11:
Hi

I have solved
In your guide there is a number of package missing
If you want I can share with you the whole list of packages I have used in
order to have a full feature working Ubuntu environment

closed 2023-10-20 by jmottier-niryo