NiryoRobotics/ned_ros #11: Robot is losing the connection after sending a command to

fabiocyro, 2022-10-17:
Hello.

I have some Ned2 arms and i am having a problem that is getting hard to solve it with one of them:

Once I connect the gripper (large) and send a command to either open or close it, everything freezes. Also, after a few seconds I get a message that the "Robot connection is closed" following by the arm stopping working and rebooting itself. The problem is that there is not even an error that it could be shown on ROS logs for example.

The arm is brand new. At first, I thought it was a problem with the gripper until I tested a brand new one and got the same error. The grippers work fine on other arms also.

Any ideas?

Jpl-niryo, 2022-11-15:
Hi,

The problem you are describing looks like a gripper overload error. The gripper motors are a little sensitive and can go into error if they cannot close completely. Does the Ned2 work fine without the gripper? You could also try reflashing the Ned2's SD card.

fabiocyro, 2022-11-15:
> Hi,
> 
> The problem you are describing looks like a gripper overload error. The gripper motors are a little sensitive and can go into error if they cannot close completely. Does the Ned2 work fine without the gripper? You could also try reflashing the Ned2's SD card.

Hello! Thanks for getting in touch.

Indeed the error is related to overload, the thing is that we are to. After performing some more tests I could see it on the logs the overload message. The thing is that is already the third gripper that we are using that is presenting this issue. The arm works fine without it. I will consider reflashing the SD card now =). Thank you!

Jpl-niryo, 2022-11-15:
My pleasure!

Unfortunately this does not surprise me that you are having trouble with the standard gripper. We did not set up a high enough overload threshold and they are being damaged by clients. In the meantime, we have added a proper software protection that will prevent gripper damage. If you would like we can send you another gripper

fabiocyro, 2022-11-15:
Oh I see, so with the protection things will get better for sure! 

That is very nice of you of proposing to send another gripper but it won't be necessary. The robots are located where I work (University in France) and we are in contact with our supplier. He told us that he would send us another gripper, two actually! He also mentioned about this software protection, so I think we are going trough the right direction.

In any case, thank you for the attention and have a very nice day!

closed 2022-11-15 by fabiocyro