NiryoRobotics/ned_ros #12: frame size

otorunoglu, 2023-02-20:
Hi,
How can we change frame size for Ned2? 
We want to use different USB camera. We can connect camera and get frames. 
But frame resolution is 640x360. Our USB camera normal resolution is 1920x1080.
We have tried to change /niryo_robot_vision/config/ned2/video_server_setup.yaml frame size. 
But it did not have the desired effect.
Can you give any ideas?

ValentinPitre, 2023-03-02:
Hi! 
With your new camera, are you able to visualize the stream?
Did you try to let the _**obj_calib_name**_ field empty. Otherwise, he will try to undistort the image with that calibration file: https://github.com/NiryoRobotics/ned_ros/blob/master/niryo_robot_vision/config/cam_intrinsics.yaml. In this file, there is also resolutions params.

otorunoglu, 2023-03-03:
Hi, thank you for your answer.

Yes we can get image from our new camera. We have already changed the parameters on the video_server_setup.yaml and cam_intrinsics.yaml files and tried to let the obj_calib_name field empty. However, the problem persists, and we can only get a resolution of 640x360.

We tried to obtain the frame from another Raspberry Pi device, and we encountered the same problem initially. However, we managed to change the camera resolution using the following code:

cap = cv2.VideoCapture(0)
cap.set(cv2.CAP_PROP_FRAME_WIDTH, 1920)
cap.set(cv2.CAP_PROP_FRAME_HEIGHT, 1080)

This resolved the issue. My question is, can we use the same system on the Pyniryo Vision codes before the code "img_compressed = robot.get_img_compressed()"? Alternatively, do you have any other suggestions that may help us resolve this problem?

Thank you for your help.

jmottier-niryo, 2023-10-20:
Hello, you can't make these modifications from pyniryo.
If you want to change the resolution, you will have to either make your own package to handle the video stream, or edit our package niryo_robot_vision manually