NiryoRobotics/ned_ros #20: import error for Camera vision

pranavrnair22, 2024-06-17:
I am unable to view the camera in gazebo , when I execute this command -- roslaunch niryo_robot_bringup desktop_gazebo_simulation.launch ; getting this error:

--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

[INFO] [1718600785.556666, 0.000000]: Programs Manager - Started
Traceback (most recent call last):
  File "/home/pn20/catkin_ws_niryo_ned/devel/lib/niryo_robot_vision/camera_publisher_and_services.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/pn20/catkin_ws_niryo_ned/src/niryo_robot_vision/scripts/camera_publisher_and_services.py", line 19, in <module>
    from ObjectDetector import ObjectDetector
ImportError: cannot import name 'ObjectDetector' from 'ObjectDetector' (/home/pn20/catkin_ws_niryo_ned/devel/lib/niryo_robot_vision/ObjectDetector.py)
[INFO] [1718600785.876388, 0.000000]: Poses Handlers - Started
[niryo_robot_vision-19] process has died [pid 45098, exit code 1, cmd /home/pn20/catkin_ws_niryo_ned/devel/lib/niryo_robot_vision/camera_publisher_and_services.py __name:=niryo_robot_vision __log:=/home/pn20/.ros/log/10a5cf3e-2c65-11ef-8a3b-49b52e7b4f50/niryo_robot_vision-19.log].
log file: /home/pn20/.ros/log/10a5cf3e-2c65-11ef-8a3b-49b52e7b4f50/niryo_robot_vision-19*.log
[INFO] [1718600786.179096, 0.000000]: Python ROS Wrapper ready
[INFO] [1718600786.215293, 0.000000]: TCP Server - Started on port 40001
[INFO] [1718600786.226496, 0.000000]: User Interface Node - Started
[INFO] [1718600786.287384, 0.000000]: Python ROS Wrapper ready
[gazebo-20] process has died [pid 45099, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/pn20/catkin_ws_niryo_ned/src/niryo_robot_gazebo/worlds/niryo_cube_world.world __name:=gazebo __log:=/home/pn20/.ros/log/10a5cf3e-2c65-11ef-8a3b-49b52e7b4f50/gazebo-20.log].
log file: /home/pn20/.ros/log/10a5cf3e-2c65-11ef-8a3b-49b52e7b4f50/gazebo-20*.log
Traceback (most recent call last):
  File "/home/pn20/catkin_ws_niryo_ned/devel/lib/niryo_robot_programs_manager_v2/programs_manager_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context

jmottier-niryo, 2024-06-17:
Hello, this is a known problem related to the command used to build the stack.
To prevent this error, please rebuild the stack using `catkin_make install` (To prevent any weird behavior, I recommend deleting the old build with `rm -rf build devel`). Once the build is complete, you will have to source the file located in `install/setup.bash` (instead of the one located in devel). After that you should be able to run the Ned stack without issue.

pranavrnair22, 2024-06-17:
Thank you for quick response, it is working now

closed 2024-06-17 by jmottier-niryo