NiryoRobotics/ned_ros #21: niryo_robot_saved_files/niryo_robot_versions/version.txt' not found

zainabsaeed24, 2024-07-24:
While running the `roslaunch niryo_robot_bringup niryo_ned2_robot.launch timed:=false`, we run into the following error and we are not able to find the directory online.

```
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/__init__.py", line 347, in main
    p.start()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 305, in start
    self._start_infrastructure()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 254, in _start_infrastructure
    self._load_config()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 156, in _load_config
    self.config = roslaunch.config.load_config_default(self.roslaunch_files, self.port,
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/config.py", line 461, in load_config_default
    loader.load(f, config, argv=args, verbose=verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 763, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 735, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 670, in _recurse_load
    self._recurse_load(ros_config, tag.childNodes, child_ns, \
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 701, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 96, in call
    return f(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 635, in _include_tag
    self._recurse_load(ros_config, launch.childNodes, child_ns, \
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 701, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 96, in call
    return f(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 635, in _include_tag
    self._recurse_load(ros_config, launch.childNodes, child_ns, \
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 670, in _recurse_load
    self._recurse_load(ros_config, tag.childNodes, child_ns, \
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 670, in _recurse_load
    self._recurse_load(ros_config, tag.childNodes, child_ns, \
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 681, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 96, in call
    return f(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 271, in _param_tag
    value = self.param_value(verbose, name, ptype, *vals)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/loader.py", line 479, in param_value
    with open(textfile, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/cpsadmin/niryo_robot_saved_files/niryo_robot_versions/version.txt'
```

jmottier-niryo, 2024-07-24:
Hello, are you trying to run the ROS stack in simulation ?
If so, you have to use the simulation launchfiles instead of niryo_ned2_robot

zainabsaeed24, 2024-07-24:
> I am running the ROS stack with Ned2 robot not in simulation.
>
> Best,
> Zainab Wattoo

My end goal is to find the base coordinates of the Ned2. Is there any way I
could find that? Any help in this regard would be appreciated. Thank you.

Best,
Zainab

jmottier-niryo, 2024-07-25:
What is your setup exactly ? Judging by the path `/home/cpsadmin` it looks like you are not running the command from the robot

What do you mean by "base coordinates" ? If you're looking for the TCP coordinates, you can get it via a topic or via PyNiryo, our python API

zainabsaeed24, 2024-07-25:
*Overall Goal:*

Our objective is to establish a global coordinate system (which could be
with respect to the TCP) for two robot arms (Ned2 and ViperX) mounted on a
platform. We plan on designating Ned2 as a reference for the global
coordinate system, with other robots that are placed on the platform using
the global coordinate system.

*Challenge:*

To achieve this, we require the base frame/world frame of the robot's
coordinates other than the TCP. To achieve this, we want to generate the
frame in ubuntu where all robots launched using ROS are shown. We are able
to launch ViperX via roslaunch command in the world coordinate system and
able to see in the world frame whereas we want to do similar with Ned2 for
viewing its frame in the global coordinate system.
We have a Ned2 robot connected to our computer via ethernet cable. We are
following the instructions posted on this link:
https://archive-docs.niryo.com/dev/ros/v4.1.1/en/source/real_robot.html
We could not find the installed repository when Ned2 was connected with our
computer, therefore, we used the following github repository (`
https://github.com/NiryoRobotics/ned_ros`) for downloading. We did
`catkin_make` and converted it into a launch file package, and are now
running the following command (roslaunch niryo_robot_bringup
niryo_ned2_robot.launch) that runs into the issue of versions.txt.
How can we confirm that the roslaunch command is specifically launching the
Ned2 robot from the command line and how to fix the issue of versions.txt?

Regarding the coordinate system, we are referring to the base frame of the
robot coordinates. The TCP coordinates are indeed accessible but the
coordinates of the base frame is something we are looking for.

If you have any other way of finding the global coordinate system where we
could make Ned2 as the point of reference, we would appreciate any help in
this regard.

Best,
Zainab

jmottier-niryo, 2024-07-26:
Our ROS stack is designed to run only on the Raspberry Pi because it needs to communicate with the hardware bus. The best approach is to run each robot independently. For more details, see [Running ROS across multiple machines](https://wiki.ros.org/ROS/Tutorials/MultipleMachines).

Regarding your question about coordinate goals, you just need to publish the transform from the Ned2's base to the ViperX's base.

To get the coordinates of the base, you can subscribe to the topic /tf, which publish all the transforms on the ned2. But I don't think you need it since the base coordinates are [0, 0, 0]

I hope this helps!

zainabsaeed24, 2024-07-30:
That makes sense. Thank you so much! This was really helpful.

Regards,
Zainab

closed 2024-07-31 by jmottier-niryo

zainabsaeed24, 2024-08-14:
Hi Niryo,

We are working on simulating the Ned2 in Gazebo (Fortress) with ROS2. We
see that you have packages for launching a simulation in Gazebo classic and
a Python ROS wrapper for control. Would there be similar software support
in the near future for the newer version(s) of Gazebo compatible with
ROS2?

Thanks and Regards,

jmottier-niryo, 2024-08-19:
Hi Zainab,

ROS2 is certainly the subject of many internal discussions. We fully recognize the importance of transitioning to ROS2 and are committed to moving in that direction. However, we currently do not have a concrete roadmap for a full ROS2 implementation.

That said, we're actively exploring possibilities to make our software ROS2-compatible.

Please feel free to reach out if you have any further questions
Justin

zainabsaeed24, 2024-08-19:
Thank you for the message. We hope to get ROS2 compatibility with Ned2 soon.

We have another unrelated question:
We would like to find the joint cartesian (x,y,z) position for all joints
whenever a robot moves to a different pose.
End Goal: We are trying to do collision checking of ned2 with any
stationary object with not only gripper position but with viperx body as
well.

Thanks and Regards,
Zainab

jmottier-niryo, 2024-08-20:
Hi Zainab,

In order to do that you'll have to compute the forward cinematic of each of the robot's joints. You can use either the /tf topic which publish the transform of all the robot's frames, or the rosparam /robot_description which contains the robot' URDF.

The later should be more efficient since the only variable is the joints angles.

On a side note, if you have any questions that aren't related to the original topic, feel free to open a new issue. :)

Justin