NiryoRobotics/ned_ros #22: Finding every joint's cartesian position (x,y,z) in relation to the base frame

zainabsaeed24, 2024-08-22:
We would like to find the joint cartesian (x,y,z) position for all joints in relation to the base frame whenever a robot moves to a different pose.

**End Goal:** We are trying to do collision checking of ned2 with any stationary object with not only gripper position but with ned2's body as well.

_**Note:**_ We want to know the joint cartesian position without using ROS and instead would like to use the python API.

jmottier-niryo, 2024-08-23:
The only way I can think of to achieve this is by calculating the forward kinematics yourself, which requires knowing the robot's description.
Unfortunately, pyniryo won't help you, as it doesn't offer functions to retrieve the joint's Cartesian positions or the robot's description.

jmottier-niryo, 2024-10-08:
Closed due to inactivity

closed 2024-10-08 by jmottier-niryo