NiryoRobotics/ned_ros #24: Issue with installation on Ubuntu 18.04 with ROS Melodic

lucaperrin, 2024-10-29:
Hi there,

I am new to ROS and interfacing with Niryo Robots. I want to be able to run simulations on the NED2 arms and followed the instruction on[ this page](https://archive-docs.niryo.com/dev/ros/v4.1.1/en/source/installation/ubuntu_18.html) following the installation of ROS Melodic on Ubuntu 18.04, as indicated on Niryo's website.

I get errors when trying to run the "catkin_make"... I have tried different things to force C++17 instead of 20 because I think C++20 is not compatible with ROS Melodic or Ubuntu 18.04 but could not figure it out.

Any help would be appreciated!
Thanks in advance.

[image]

tdgxNiryo, 2024-10-29:
Hey @lucaperrin !

The current version of the robotic stack, which is running ROS Noetic, needs to be deployed on a device running ubuntu 20.04.
If you really wish to install an older version of the stack with support for ROS Melodic, you can instead clone the version 4.1.1 which was targeting ROS Melodic (https://github.com/NiryoRobotics/ned_ros/tree/v4.1.1).

Note that in order to have access to the last upgrades of the stack, we recommend you to switch to ubuntu 20.04 in order to be able to build the latest version.

I hope this helps you !

lucaperrin, 2024-10-29:
Thanks for the quick reply!
Just to confirm, I need ROS Noetic on Ubuntu 20.04 in order to run the NED ROS simulation ?

I guess the instructions on [this page](https://archive-docs.niryo.com/dev/ros/v4.1.1/en/source/installation/ubuntu_18.html) are not up-to-date then ? Since they mention that Ubutnu 18.04 with ROS Melodic is the only currently supported OS.

lucaperrin, 2024-10-29:
Sounds good, thank you so much!

closed 2024-10-29 by lucaperrin